dusty-nv/jetson-utils #159: cmake error  /usr/bin/ld: cannot find -lnvbuf_utils

kankanjiuzou123, 2023-05-09:
In the process of compiling jetson_utils make install on the jetson orin device, I encountered the error /usr/bin/ld: cannot find -lnvbuf_utils. What is the reason and how to solve it?

kankanjiuzou123, 2023-05-09:
When I changed enable_nvmm_default to OFF, the compilation was successful, but I don't know why this happens。

kankanjiuzou123, 2023-05-10:
> @kankanjiuzou123 do you have `libnvbuf_utils.so` under `/usr/lib/aarch64-linux-gnu/tegra`?

yes
[image]

dusty-nv, 2023-05-10:
OK, does appear that you have it, and that directory is added to the linker path here:   https://github.com/dusty-nv/jetson-utils/blob/0bcb19b498326eb866a80d7d13388b2e59bc9dfd/CMakeLists.txt#L92

So I'm not sure why it's not finding it... I'm curious, if you build jetson-inference instead (which has jetson-utils as a submodule), does it find it?

kankanjiuzou123, 2023-05-11:
> OK, does appear that you have it, and that directory is added to the linker path here:
>
> https://github.com/dusty-nv/jetson-utils/blob/0bcb19b498326eb866a80d7d13388b2e59bc9dfd/CMakeLists.txt#L92
>
> So [...]

thank you very much，I will try to compile jetson_inference