dusty-nv/jetson-utils #160: How to specify the decoder format

kankanjiuzou123, 2023-06-14:
Is it possible to specify the decoder format when jetson_utils gets the rtsp camera, such as h264, h265, and if so, how should I do it (python code)？

kankanjiuzou123, 2023-06-14:
I encountered an error as shown below, I wonder if it is a camera encoding problem
[image]

dusty-nv, 2023-06-14:
You can set it with the `--input-codec=h264` command-line option, or via the options dict in Python:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#python

It should match the codec that the RTSP stream already uses though

kankanjiuzou123, 2023-06-15:
ok,I will try this，thank you very much！

topherbuckley, 2024-04-18:
Let me know if I should open a separate issue, but I thought it was related enough:

I'd like to know how to find the exhaustive list of options available for this dict. I'm not seeing where this dict is actually parsed in the code. For example, how do I use `headless`? Should it be {'headless': True} or how can I find the proper syntax/naming for all these options?

topherbuckley, 2024-04-18:
Also noting that specifying h265 results in a video that cannot be opened by vlc or the system viewer. i.e.

This produces a working video:

```python
input = videoSource("csi://0")
output = videoOutput("my_video.mp4", options={"codec": "h264"})
```

Whereas this does not:
```python
input = videoSource("csi://0")
output = videoOutput("my_video.mp4", options={"codec": "h265"})
```

This is the code common to both (derived from your video-viewer.py, but using threading as I'm also trying to run a flask app beside it). 

```python
def video_loop():
    numFrames = 0
    while numFrames < 200:
        # capture the next image
        img = input.Capture()

        if img is None:  # timeout
            continue

        if numFrames % 25 == 0 or numFrames < 15:
            Log.Verbose(
                f"video-viewer:  captured {numFrames} frames ({img.width} x {img.height})"
            )

        numFrames += 1

        # render the image
        output.Render(img)

        # update the title bar
        output.SetStatus(
            "Video Viewer | {:d}x{:d} | {:.1f} FPS".format(
                img.width, img.height, output.GetFrameRate()
            )
        )

        # exit on input/output EOS
        if not input.IsStreaming() or not output.IsStreaming():
            break
    input.Close()
    output.Close()

if __name__ == "__main__":
    x = threading.Thread(target=video_loop)
    x.start()
```