dusty-nv/jetson-utils #164: impossible to use openCV with jetson_utils ?

lweingart, 2023-08-05:
Hello,

I'm using detectnet.py as a basis for my code, but I realise it's impossible to use opencv with it, or more precisely with jetson_utils.
Here is my problem, if I type `import cv2` after importing jetson_inference and jetson_utils, I get this error:

```
Traceback (most recent call last):
  File "main.py", line 3, in <module>
    from utils import draw_box
  File "/home/jetson/git/micr_v2/utils.py", line 2, in <module>
    import cv2
  File "/usr/lib/python3/dist-packages/cv2/__init__.py", line 180, in <module>
    bootstrap()
  File "/usr/lib/python3/dist-packages/cv2/__init__.py", line 152, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: /lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block
```

If I place `import cv2` before these two imports it goes well, until I run the code and end up with the following errors when reaching the line `output.Render(img)` (I only paste the end of the logs first, then below the full logs)

```
[gstreamer] gstreamer message qos ==> decoder
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstBufferManager -- map buffer size was less than max size (1382400 vs 1382407)
[gstreamer] gstBufferManager recieve caps:  video/x-raw, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)0/1, format=(string)NV12
[gstreamer] gstBufferManager -- recieved first frame, codec=H264 format=nv12 width=1280 height=720 size=1382407
[cuda]   allocated 4 ring buffers (1382407 bytes each, 5529628 bytes total)
[cuda]   allocated 4 ring buffers (8 bytes each, 32 bytes total)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[cuda]   allocated 4 ring buffers (2764800 bytes each, 11059200 bytes total)
[OpenGL] glDisplay -- set the window size to 1280x720
[OpenGL] creating 1280x720 texture (GL_RGB8 format, 2764800 bytes)
[cuda]   cudaGraphicsGLRegisterBuffer(&interop, allocDMA(type), cudaGraphicsRegisterFlagsFromGL(flags))
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/jetson/git/jetson-utils/display/glTexture.cpp:360
[cuda]   allocated 2 ring buffers (1382400 bytes each, 2764800 bytes total)
[cuda]   cudaGetLastError()
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/jetson/git/jetson-utils/cuda/cudaYUV-YV12.cu:257
[cuda]   cudaRGBToI420((uchar3*)input, output, width, height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/jetson/git/jetson-utils/cuda/cudaColorspace.cpp:128
[cuda]   cudaConvertColor(image, format, nextYUV, IMAGE_I420, width, height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/jetson/git/jetson-utils/codec/gstEncoder.cpp:703
[gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f
Traceback (most recent call last):
  File "main.py", line 80, in <module>
    output.Render(img)
Exception: jetson.utils -- videoOutput failed to render image
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstDecoder -- pipeline stopped
```

and here is the full output:
```
python3 main.py test.mp4
Prepare network card...
Running sudo ifconfig eth0 192.168.9.3 netmask 255.255.255.0 up
[... 347 lines cut ...]
[cuda]      /home/jetson/git/jetson-utils/cuda/cudaYUV-YV12.cu:257
[cuda]   cudaRGBToI420((uchar3*)input, output, width, height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/jetson/git/jetson-utils/cuda/cudaColorspace.cpp:128
[cuda]   cudaConvertColor(image, format, nextYUV, IMAGE_I420, width, height)
[cuda]      invalid OpenGL or DirectX context (error 219) (hex 0xDB)
[cuda]      /home/jetson/git/jetson-utils/codec/gstEncoder.cpp:703
[gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f
Traceback (most recent call last):
  File "main.py", line 80, in <module>
    output.Render(img)
Exception: jetson.utils -- videoOutput failed to render image
[gstreamer] gstDecoder -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstDecoder -- pipeline stopped
```

Any idea what could be done? I really need to use opencv in my code.
Any help would be deeply appreciated.

Cheers

dusty-nv, 2023-08-05:
Hi @lweingart, try doing `export LD_PRELOAD=/lib/aarch64-linux-gnu/libgomp.so.1` first

And try doing `import cv2` after you create the `videoOutput` interface - I realize this is decidedly un-Pythonic, but cv2 module likes to overwrite the OpenGL extensions...

lweingart, 2023-08-06:
Hi @dusty-nv 

Thank you very much for your help!
It indeed appears to fix the issue

Thanks again, have a great day

closed 2023-08-06 by lweingart