dusty-nv/jetson-utils #166: wbmode settings in nvarguscamerasrc

unpickme, 2023-09-11:
Hi Dusty,

I am configuring the Gstreamer pipeline for a capture device based on imx219 sensors. 
I am interested in the GST_NVCAM_WB_MODE_MANUAL (9) mode mentioned in the "gst-inspect-1.0 nvarguscamerasrc" document. This is because none of the other white balance modes seems to be exactly what we use, which is constant 4000k. colour temperature light. Are there any further details available on how you set the white balance manually if you make the 'manual' selection?

Any advice or references would be most gratefully received, many thanks in advance

dusty-nv, 2023-09-11:
@unpickme presuming you are asking about the nvarguscamerasrc settings themselves, I would post this question to the Jetson developer forums

unpickme, 2023-09-11:
Okay Dusty, thanks for getting back so promptly, I will do that.I just saw your response to an earlier nvarguscamerasrc question on jetson-utils and thought you might know more.
______________________________________
peter darrell  (FOUNDER)

4Ax Technologies
BETTER WORLDS, STARTING WITH THIS ONE

www.peterddarrell.com
https://www.linkedin.com/in/peter-darrell-4761181a/
www.peterddarrell.com/4ax