dusty-nv/jetson-utils #169: Deprecated Nvbuf_utils is removed from JetPack 5.1.2

atinfinity, 2023-09-22:
I build `jetson-utils` on JetPack 5.1.2.
As a result, CMake could not find `nvbuf_utils`.

```cmake
-- distro ID:       Ubuntu
-- distro version:  20.04
-- distro codename: focal
-- nvbuf_utils:  NVBUF_UTILS-NOTFOUND
-- NVMM zero-copy memory:  ENABLE_NVMM=OFF
```

I found the following information in <https://docs.nvidia.com/jetson/archives/jetpack-archived/jetpack-512/release-notes/index.html>.

> Deprecated Nvbuf_utils is removed. Refer to migration guide to [migrate from Nvbuf_utils to NvUtils](https://developer.nvidia.com/sites/default/files/akamai/embedded/nvbuf_utils_to_nvutils_migration_guide.pdf).

atinfinity, 2023-09-22:
@dusty-nv Yes. I was able to build it without error.

poett1, 2024-03-18:
@dusty-nv wouldn't this decrease the performance with nvarguscamerasrc on Jetpack >= 5.0? It seems like [gstBufferManager.cpp](https://github.com/dusty-nv/jetson-utils/blob/master/codec/gstBufferManager.cpp#L35C1-L35C41) already has the changes needed for Jetpack 5, is the only missing step an update of CMakeLists.txt according to [Nvbuf_utils to NvUtils Migration guide](https://developer.nvidia.com/sites/default/files/akamai/embedded/nvbuf_utils_to_nvutils_migration_guide.pdf)?

dusty-nv, 2024-03-18:
@poett1 I believe there are API changes, it wasn't just the name of the header.  I haven't really dug into it, but if you wanted to investigate it further and make a working PR that would be much appreciated 👍

poett1, 2024-04-24:
@dusty-nv i just created a [PR](https://github.com/dusty-nv/jetson-utils/pull/204). Feel free to check this out @atinfinity.