dusty-nv/jetson-utils #170: Utilizing jetson-utils for Yolov5 inference on Jetson device

NishaniKasineshan, 2023-09-25:
I am currently working on Jetson Tx2. Since I am interested in using yolov5 for jetson inference which is not supported with your jetson-inference repo I alternatively tried [https://github.com/mailrocketsystems/JetsonYolov5](https://github.com/mailrocketsystems/JetsonYolov5). Since I am using opencv for video output there is bit lag in inferencing. so i would like to utilize  jetson-utils for inference. But I am getting following error.

------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1280x720
[OpenGL] glDisplay -- X window resolution:    1280x720
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  43
  Current serial number in output stream:  42
-------------------------------------------------------------------
PyCUDA ERROR: The context stack was not empty upon module cleanup.
-------------------------------------------------------------------
A context was still active when the context stack was being
cleaned up. At this point in our execution, CUDA may already
have been deinitialized, so there is no way we can finish
cleanly. The program will be aborted now.
Use Context.pop() to avoid this problem.
-------------------------------------------------------------------
Aborted (core dumped)

any suggestions regarding how to apply jetson-utils for other models is appreciable....

dusty-nv, 2023-09-25:
> libGL error: failed to load driver: swrast

@NishaniKasineshan it appears that your NVIDIA OpenGL driver got uninstalled or another GPU driver was installed over top, otherwise this would report `nvidia`.   Are you able to run the `video-viewer` utility regularly just as a test?

NishaniKasineshan, 2023-09-26:
@dusty-nv yes the video-viewer works fine. The thing I could utilize jetson utils normally but when the yolo engine model I created using @mailrocketsystems [https://github.com/mailrocketsystems/JetsonYolov5](https://github.com/mailrocketsystems/JetsonYolov5) is loading is causing the issue.

dusty-nv, 2023-09-26:
@NishaniKasineshan in the event of a package conflict corrupting your OpenGL bindings, try importing jetson_utils and creating your videoSource/videoOutput interfaces before you import/create the YOLOv5 engine