dusty-nv/jetson-utils #171: A question about how to impl "--input-save" in source code

ithanwu, 2023-10-21:
Thanks my dude , you really made a great project ,your codes helped me a lot ,I can't say more thx.
When I read your document about the input save parts ,I found it quit valuable for me , I want to run my DL model ,and record video files at the same time ,as you know jetson orin nano has limited resource , when my modle run , the GPU raise up to 100% , and the cpus are weak too.
So ,when I saw your  --input-save ,it shined me,that's what I need ! I want to use it in my source code ,then I read the video-viewer.cpp ,but I found in your code this could only be impl as a second dest ,that's not so good ,I want to  save the V4l2 camera data to hard drive before they are  decoded ,and only save them without any other output streams 
Could you please show me how to do this ?
best regrads!

dusty-nv, 2023-10-21:
Hi @ithanwu, thank you - `--input-save` should work with V4L2 camera, as long as it is using a compressed format already.   `--input-save` will not encode the data if it is coming in raw.  But if your V4L2 camera already uses H264/mjpeg/ect, it would dump incoming, unprocessed original compressed data to disk.

dusty-nv, 2023-10-21:
Also, you can always just create a 2nd videoOutput object and re-encode it if you need to

ithanwu, 2023-10-22:
yeah , It's amazing for your fast response ,
and my v4l2 camera use mjpg ，I successfully created a video that has a mjpg codec, my problem is ,the video-videwer create the `--input-save` output video files as a second destination , there is a primary output ,I want only create the `--input-save` video file without outputing the data to any other streams ,
for example ,in the default satuation ,the video-viewer would open an OpenGL window,which could be set off by passing a `--headless` to the video-viewer,
I want to do this in my c++ code : only generate an `--input-save`  like video ,without generating a so called primary output stream.
I mean I have already found an videoOptions memeber .save ,that's related to the  `--input-save` or `--output-save`,I want to know how to generate an `--input-save` videofiles without generating additional streams like display stream or network streams ,only the `--input-save` result