dusty-nv/jetson-utils #177: Best way to load binary image for object detection

peltiern, 2023-11-12:
Hi @dusty-nv,

What is the best way to load an image received in binary format to perform object detection with jetson-inference (python grpc server) ?

Currently I save the binary image to a temporary file and use jetson_utils.loadImage() method to convert it to CUDA. 
The image saving and loading time is around **48ms** and the total time (saving + loading + detection) is around **98ms**. So I spend almost half the time saving + loading the image.
Do you know another more effective method?

Here is my code :

```
def detect_objects(self, image_data):
        before = int(round(time.time() * 1000))
        # Enregistrez le tableau de bytes dans un fichier temporaire
        temp_file_path = "/jetson-inference/temp_image.jpg"

        # Save the bytes to the temporary file
        with open(temp_file_path, "wb") as temp_file:
            temp_file.write(image_data.read())

        # Chargez l'image à partir du fichier temporaire
        img_cuda = loadImage(temp_file_path)
        
        after = int(round(time.time() * 1000))
        print("temps chargement image : " + str(after - before) + " ms")

        return self.net.Detect(img_cuda, overlay="box,labels")
```

Best regards.

Nicolas

dusty-nv, 2023-11-13:
Hi @peltiern, what is the format / datatype of your image?   You can either map a jetson_utils.cudaImage from an existing ptr, or allocate a cudaImage and copy to it.  See here for more info:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#sharing-the-memory-pointer

peltiern, 2023-11-14:
Hi @dusty-nv ,

When the image is loaded, this is displayed: `[image]  loaded '/jetson-inference/temp_image.jpg'  (640x480, 3 channels)`

With the **identify** util :
```
jetson@Wall-E:~$ identify -verbose temp_image.jpg 
Image: temp_image.jpg
  Format: JPEG (Joint Photographic Experts Group JFIF format)
  Mime type: image/jpeg
  Class: DirectClass
  Geometry: 640x480+0+0
  Units: Undefined
  Type: TrueColor
  Endianess: Undefined
  Colorspace: sRGB
  Depth: 8-bit
  Channel depth:
    red: 8-bit
    green: 8-bit
    blue: 8-bit
  Channel statistics:
    Pixels: 307200
    Red:
      min: 0 (0)
      max: 150 (0.588235)
      mean: 48.0959 (0.188611)
      standard deviation: 33.9316 (0.133065)
      kurtosis: -1.16359
      skewness: 0.231503
      entropy: 0.898698
    Green:
      min: 0 (0)
      max: 150 (0.588235)
      mean: 55.5244 (0.217743)
      standard deviation: 40.4401 (0.158588)
      kurtosis: -1.03826
      skewness: 0.346835
      entropy: 0.929196
    Blue:
      min: 0 (0)
      max: 165 (0.647059)
      mean: 54.4834 (0.21366)
      standard deviation: 43.7808 (0.171689)
      kurtosis: -0.923883
      skewness: 0.580087
      entropy: 0.92195
  Image statistics:
    Overall:
      min: 0 (0)
      max: 165 (0.647059)
      mean: 52.7012 (0.206671)
      standard deviation: 39.5959 (0.155278)
      kurtosis: -0.828971
      skewness: 0.476718
      entropy: 0.916615
  Rendering intent: Perceptual
  Gamma: 0.454545
  Chromaticity:
    red primary: (0.64,0.33)
    green primary: (0.3,0.6)
    blue primary: (0.15,0.06)
    white point: (0.3127,0.329)
  Background color: white
  Border color: srgb(223,223,223)
  Matte color: grey74
  Transparent color: black
  Interlace: None
  Intensity: Undefined
  Compose: Over
  Page geometry: 640x480+0+0
  Dispose: Undefined
  Iterations: 0
  Compression: JPEG
  Quality: 95
  Orientation: Undefined
  Properties:
    date:create: 2023-11-14T16:59:13+01:00
    date:modify: 2023-11-14T16:58:55+01:00
    jpeg:colorspace: 2
    jpeg:sampling-factor: 2x2,1x1,1x1
    signature: 704e71f3598082946bc27d9dd78395b092b8bd3cce78f5e5400bc905fecc81b9
  Artifacts:
    filename: temp_image.jpg
    verbose: true
  Tainted: False
  Filesize: 69.9KB
  Number pixels: 307K
  Pixels per second: 15.36MB
  User time: 0.010u
  Elapsed time: 0:01.019
  Version: ImageMagick 6.9.7-4 Q16 aarch64 20170114 http://www.imagemagick.org
```
So I guess it's the **IMAGE_RGB8** format.

I don't understand how to associate the databytes with the ptr.

Nicolas

dusty-nv, 2023-11-14:
@peltiern what kind of Python object is `image_data` in your code?

peltiern, 2023-11-14:
@dusty-nv 

I send byte array of the image from a Java client (from webcam source) to a Python server via gRPC.

In server side, the byte array is decoded with `io.BytesIO(request.image)`.

Here is the Python [server code](https://github.com/peltiern/robot/blob/ab232c59db818b69c4444b861ce47bf0451b4155/robot-python/vision-artificielle/serveur-python/ai-server-grpc.py#L37) and the [object detector code](https://github.com/peltiern/robot/blob/ab232c59db818b69c4444b861ce47bf0451b4155/robot-python/vision-artificielle/serveur-python/object/jetson_inference/jetson_object_detector.py#L13)

peltiern, 2023-11-19:
Hi @dusty-nv ,

I re-tested and I can't resolve the issue when using the **cudaMemcpy()** function : **jetson.utils -- failed to get CUDA image from src argument**

Here is my code with two different functions: one using the save to temporary file and the **loadImage()** function and the other using the **cudaMemcpy()** function. The input parameter type is always the same: **_bytes_**

```python
    def detect_objects(self, image_data):

        return self.detect_objects_with_cudaMemcpy(image_data)

    def detect_objects_with_temp_file(self, image_data):

        before = int(round(time.time() * 1000))

        temp_file_path = "/vision-artificielle/temp_image.jpg"

        # Save the bytes to the temporary file
        with open(temp_file_path, "wb") as temp_file:
            temp_file.write(image_data)
        # Load CUDA image from temp file
        img_cuda = jetson_utils.loadImage(temp_file_path)

        after = int(round(time.time() * 1000))
        print("Image load time : " + str(after - before) + " ms")

        return self.net.Detect(img_cuda, overlay="box,labels")

    def detect_objects_with_cudaMemcpy(self, image_data):
        image_data_type = type(image_data)
        print(f"Type image_data = {image_data_type}")

        before = int(round(time.time() * 1000))
        
        print("Before cudaMemcpy")
        try:
            img_cuda = jetson_utils.cudaMemcpy(image_data)
            print("After cudaMemcpy")
        
            after = int(round(time.time() * 1000))
            print("Image load time : " + str(after - before) + " ms")

            return self.net.Detect(img_cuda, overlay="box,labels")
        except Exception as e:
            print(f"Error : {e}")
```

The first function with **loadImage()** works fine : 
```
[image]  loaded '/vision-artificielle/temp_image.jpg'  (640x480, 3 channels)
Image load time : 47 ms
```

The second function with **cudaMemcpy()** still has the same error:
```
Type image_data = <class 'bytes'>
Type image_data_bytes = <class 'bytes'>
Before cudaMemcpy
Error : jetson.utils -- failed to get CUDA image from src argument
```

Do you have any idea of the problem ?

Best regards.
Nicolas

dusty-nv, 2023-11-19:
@peltiern considering your bytes memory is allocated on the CPU, I would try using [`numpy.frombuffer()`](https://numpy.org/doc/stable/reference/generated/numpy.frombuffer.html), then [`numpy.reshape()`](https://numpy.org/doc/stable/reference/generated/numpy.reshape.html) to give it the correct image dimensions, and then [`jetson_utils.cudaFromNumpy()`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-from-numpy-arrays)

peltiern, 2023-11-21:
Hi @dusty-nv

Thank you for all the time you give me.

Here is the method I implemented using `frombuffer()` and `cudaFromNumpy()` :

```python
    def detect_objects_with_numpy(self, image_data):

        before = int(round(time.time() * 1000))

        array = np.frombuffer(image_data, dtype = np.uint64)
        img_cuda = jetson_utils.cudaFromNumpy(array, isBGR=False)

        after = int(round(time.time() * 1000))
        print("Image load time : " + str(after - before) + " ms")

        return self.net.Detect(img_cuda, width=640, height=480, overlay="box,labels")
```

I have just tested several possible combinations in terms of the values of _**dtype**_ and **_isBGR_**.
Combinations that do not crash are :
```
dtype=uint8 and isBGR=True
dtype=uint8 and isBGR=False
```
But unfortunately, the detector does not find any results unlike the method using `loadImage()`.

**_image_data_** is an array of bytes of an image of 640*480 (see attachment which corresponds to the temporary image when I use `loadImage()`).

Can the `reshape()` method change anything ?
Or do you have another idea? I feel like I'm getting closer to the goal.

Otherwise, I'll just use the `loadImage()` method.

Best regards,
Nicolas

[image]

dusty-nv, 2023-11-21:
Hi Nicolas, `np.reshape()` doesn't change anything, it just sets the array dims properly.  Try it like this:

```python
array = np.frombuffer(image_data, dtype = np.uint8)
array = np.reshape(array, (480, 640, 3))

img_cuda = jetson_utils.cudaFromNumpy(array, isBGR=True)
jetson_utils.saveImage('test.jpg', img_cuda)
```

Then check `test.jpg` and see if the conversion was successful.

peltiern, 2023-11-23:
Hi @dusty-nv ,

I failed to use the `reshape` method. I get the error "_cannot reshape array of size 79837 into shape (480,640,3)_". Which I think is normal because `image_data` is a bytearray of a JPEG image, and therefore has a variable size for each image (even if the dimensions are the same).

I managed, after multiple errors/retry/... to find a solution using PIL (thanks ChatGPT ;-)). Why does it work going this way ? I don't know ...

Now, the detection endpoint takes around **53ms** for a 640*480 image, instead of **98ms** with writing a temporary image ! Almost **20 FPS**.

Thank you very much for your help and your time. We can close the ticket unless you have other solutions/explanations.

Nicolas.

Here is the code : 
```python
    def detect_objects(self, image_data):
        
        # Convert databytes to PIL image
        image = Image.open(io.BytesIO(image_data))
        # Convert PIL image to NumPy array
        array = np.array(image)
        # Convert Numpy array to CUDA image
        img_cuda = jetson_utils.cudaFromNumpy(array, isBGR=False)

        # Detection
        return self.net.Detect(img_cuda, width=640, height=480, overlay="box,labels")
```

dusty-nv, 2023-11-26:
> Which I think is normal because `image_data` is a bytearray of a JPEG image
>  Why does it work going this way ? 

Oh, I had thought it was an RGB array - it works with PIL, because it probably recognizes the JPEG byte header and then decompresses it.  Either way, glad you got it working!

closed 2023-11-26 by dusty-nv