dusty-nv/jetson-utils #180: Potential memory leak in RingBuffer

SeeRich, 2023-11-17:
When I run the following code, Valgrind reports that there was 300 bytes leaked from malloc:
```
auto rb = RingBuffer();
rb.Alloc(3, 100);
```

I took a look at the code in question (`RingBuffer.inl`) and I think the malloc'd memory pointer is being overwritten by a call to `cudaMalloc`?
```
mBuffers[n] = malloc(size);

if( CUDA_FAILED(cudaMalloc(&mBuffers[n], size)) )
{
    LogError(LOG_CUDA "RingBuffer -- failed to allocate CUDA buffer of %zu bytes\n", size);
    return false;
}
```

It looks like `cudaFree` is being called on each of the buffers but not `free`. If the ring buffer is only designed to allocate GPU/Mapped memory, is this `malloc` even needed?

dusty-nv, 2023-11-17:
Ahh, thanks @SeeRich, yes this would appear to be a bug.  If you remove `mBuffers[n] = malloc(size);` and recompile, do you get any runtime errors?

SeeRich, 2023-11-17:
@dusty-nv, yes. If I comment out that line, the Valgrind report is clean.

SeeRich, 2023-11-17:
Just from searching the code, it doesn't look like this is being used. `RingBuffer` seems to only be used in `gstBufferManager.cpp` and `gstEncoder.cpp` and both use cases specify `RingBuffer::ZeroCopy` which would bypass this issue. On the other hand, if a consumer of this project was using non-`ZeroCopy`, they would probably see a performance gain (and lower memory usage).

dusty-nv, 2023-11-17:
Thanks @SeeRich!, yes agreed 👍 

Patched this in https://github.com/dusty-nv/jetson-utils/commit/d374d425e332f5732204407fdaba8cc271e376ea and updated the submodule in jetson-inference in https://github.com/dusty-nv/jetson-inference/commit/aa91738ffa51d2b99ccae495e219039a408ba136

closed 2023-11-17 by SeeRich