dusty-nv/jetson-utils #185: WebRTC streaming failure on TX2/L4T-32.7.4

Hongfeng-mtec, 2023-11-29:
when using `./video-viewer /dev/video1 webrtc://@:8554/cam0 --headless`  the video-viewer will crash with following error:
```
video-viewer:  captured 475 frames (1280x720)
[webrtc] websocket /cam0 -- new connection opened by 192.168.55.100 (peer_id=0)
[webrtc] new WebRTC peer connecting (192.168.55.100, peer_id=0)
(video-viewer:11761): GLib-GObject-WARNING **: 19:21:59.030: g_object_set_is_valid_property: object class 'GstWebRTCBin' has no property named 'latency'
ERROR:/jetson-utils/codec/gstEncoder.cpp:866:static void gstEncoder::onWebsocketMessage(WebRTCPeer*, const char*, size_t, void*): 'sinkpad' should not be NULL
Aborted (core dumped)
```

Tried CSI camera with same result.
Setup: Jetson TX2 dev-kit freshly flashed with JP4.6.4, Jetson-utils compiled from source code.
Browser: Safari 17.0, also tried Chrome with #enable-webrtc-hide-local-ips-with-mdns disabled, yielding same result.

I had this working several months ago when I was using L4T32.7.3, but now I can't make it work even if I revert back to the old L4T version. Please help me figure out what's broken here. Any guidance is greatly appreciated.

Hongfeng-mtec, 2023-11-29:
Update: after installing `gstreamer1.0-nice` I no longer see the NULL pointer issue. However many times I see such error
```
[network] getHostByName() unable to resolve host '0287ad3d-b146-4d71-9d3e-ded491cc8822.local'
[webrtc] couldn't resolve 0287ad3d-b146-4d71-9d3e-ded491cc8822.local from SDP candidate string
```
How to fix this from the Jetson side? 
FYI: I am using Safari17

Hongfeng-mtec, 2023-12-04:
For those who have similar issues: I found that installing avahi-daemon fixed the problem.

closed 2023-12-04 by Hongfeng-mtec

dusty-nv, 2023-12-04:
Thanks @Hongfeng-mtec, appreciate you posting your solution - I will make a note to investigate installing `avahi-daemon` when these mDNS issues occur