dusty-nv/jetson-utils #186: CMake Error at python/bindings_python_3.6/cmake_install.cmake:89 (file):

zjykzj, 2023-11-30:
Running on NVIDIA Xavier, Jetpack [L4T 32.5.2]. I compiled `jetson-utils` in container `nvcr.io/nvidia/l4t-pytorch:r32.5.0-pth1.7-py3`, Follow the introduction of README:

```text
git clone https://github.com/dusty-nv/jetson-utils
mkdir build
cd build
cmake ../
make -j$(nproc)
sudo make install
sudo ldconfig
```

And use [jetson-inference/CMakePreBuild.sh](https://github.com/dusty-nv/jetson-inference/blob/master/CMakePreBuild.sh) to install dependencies.

## QUESTION1

Get ERROR when `make -j$(nproc)`

```text
In file included from /workdir/jetson-utils/codec/gstWebRTC.cpp:23:0:
/workdir/jetson-utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
 #include <gst/webrtc/webrtc.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/jetson-utils.dir/build.make:4403: recipe for target 'CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-utils.dir/codec/gstWebRTC.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /workdir/jetson-utils/codec/gstEncoder.cpp:24:0:
/workdir/jetson-utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
 #include <gst/webrtc/webrtc.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/jetson-utils.dir/build.make:4355: recipe for target 'CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o] Error 1
In file included from /workdir/jetson-utils/codec/gstDecoder.cpp:24:0:
/workdir/jetson-utils/codec/gstWebRTC.h:30:10: fatal error: gst/webrtc/webrtc.h: No such file or directory
 #include <gst/webrtc/webrtc.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/jetson-utils.dir/build.make:4331: recipe for target 'CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

Fix it refer to [Missing dependencies when building on 35.1.0 container images possible issues with opencv #146](https://github.com/dusty-nv/jetson-utils/issues/146)

```shell
apt-get install libgstreamer1.0-dev
apt-get install libjson-glib-dev
apt-get install libgstreamer-plugins-bad1.0-dev
```

## QUESTION2

Get ERROR when `make install`

```text
root@localhost:/workdir/jetson-utils/build_v2# make install
[ 83%] Built target jetson-utils
[ 97%] Built target jetson-utils-python-36
[100%] Built target video-viewer
Install the project...
-- Install configuration: ""
-- Up-to-date: /workdir/jetson-utils/install/include/jetson-utils/alphanum.h
-- Up-to-date: /workdir/jetson-utils/install/include/jetson-utils/commandLine.h
-- Up-to-date: /workdir/jetson-utils/install/include/jetson-utils/csvReader.h
-- Up-to-date: /workdir/jetson-utils/install/include/jetson-utils/csvWriter.h
[... 82 lines cut ...]
-- Up-to-date: /workdir/jetson-utils/install/bin/cuda-to-cv.py
-- Up-to-date: /workdir/jetson-utils/install/bin/cuda-to-numpy.py
-- Up-to-date: /workdir/jetson-utils/install/bin/cuda-to-pytorch.py
-- Up-to-date: /workdir/jetson-utils/install/bin/test-display.py
-- Up-to-date: /workdir/jetson-utils/install/bin/test-logging.py
-- Up-to-date: /workdir/jetson-utils/install/bin/test-video.py
-- Up-to-date: /workdir/jetson-utils/install/bin/video-viewer.py
-- Up-to-date: /workdir/jetson-utils/install/bin/test-cuda.sh
-- Up-to-date: /usr/lib/python3.6/dist-packages/jetson_utils_python.so
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/__init__.py
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/Utils
-- Up-to-date: /usr/lib/python3.6/dist-packages/Jetson/Utils/__init__.py
CMake Error at python/bindings_python_3.6/cmake_install.cmake:89 (file):
  file INSTALL cannot find
  "/workdir/jetson-utils/python/bindings/../python/jetson".
Call Stack (most recent call first):
  python/cmake_install.cmake:94 (include)
  cmake_install.cmake:393 (include)

Makefile:117: recipe for target 'install' failed
make: *** [install] Error 1
```

How can I fix it? @dusty-nv Looking forward to your answer

dusty-nv, 2023-11-30:
Hi @zjykzj, jetson-utils has a container and Dockerfile for it here:   https://github.com/dusty-nv/jetson-containers/tree/master/packages/jetson-utils

zjykzj, 2023-11-30:
> Hi @zjykzj, jetson-utils has a container and Dockerfile for it here: https://github.com/dusty-nv/jetson-containers/tree/master/packages/jetson-utils

This document shows dependency environment `L4T>=32.6` and My environment is `32.5.2`. I tried [dusty-nv/jetson-inference](https://github.com/dusty-nv/jetson-inference) 's container: https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md 

```text
$ git clone --recursive --depth=1 https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference
$ docker/run.sh
```

and run the examples in `python/examples/cuda-from-numpy.py`, that works. Just need to modify some things:

```
# from jetson_utils import cudaFromNumpy, saveImageRGBA
from jetson.utils import cudaFromNumpy, saveImageRGBA
```

dusty-nv, 2023-11-30:
OK cool.  It says dependency `L4T>=32.6` because JetPack 4.6 is as old as I support in jetson-containers, but it should still be okay.