dusty-nv/jetson-utils #189: videoOutput.Render() save video error

kankanjiuzou123, 2023-12-07:
I use videoOput.Render to save the video, and the following error occurs. However, I have no problem using the same code on other orin devices. What is the reason?
[image]
[image]

kankanjiuzou123, 2023-12-07:
[image]
I'm also very confused here, why need to create gldisplay to save the video?

dusty-nv, 2023-12-07:
> The difference is that after reporting the error, vscode ssh can save the video, while mobaxterm will cause the program to interrupt. What is the reason for this? I look forward to your reply.

Hi @kankanjiuzou123, in your MobaXterm session settings, make sure `X11-Forwarding` is disabled.   jetson-utils uses CUDA/OpenGL interoperability for display, and that doesn't work with X11 tunneling.

dusty-nv, 2023-12-08:
Hi @kankanjiuzou123, I'm not exactly sure what's going on or if there is some other setting in MobaXTerm (it seems you have a newer version than I do), but due to the `cudaGraphicsGLRegisterBuffer()` error it does point to an issue with the display.  

Try running video-viewer/detectnet/ect program with `--headless` flag and see if that helps, it will disable use of the display.