dusty-nv/jetson-utils #190: Webrtc server with GStremer webrtcsrc client

SeeRich, 2024-01-13:
I would like to run the webrtc streaming server from this library and receive the stream in a client application using gstreamer’s webrtcsrc here:

https://gstreamer.freedesktop.org/documentation/rswebrtc/webrtcsrc.html?gi-language=c

Is this possible? If so, where do I get the peer ID?

dusty-nv, 2024-01-13:
> I would like to run the webrtc streaming server from this library and receive the stream in a client application using gstreamer’s webrtcsrc

@SeeRich I'm not familiar with webrtcsrc/webrtcsink, are they newer than webrtcbin?   That's what I use, and there is a lot of code for negotiation.  It does work to both send/receive with clients using standard javascript however.  And you can use it with gstreamer for both send/recieve like here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#webrtc

> If so, where do I get the peer ID?

Correct me if I'm wrong, but those plugins don't expose a peer ID and the concept of a peer ID is more at the level of the matchmaking signalling server.  With jetson_utils, you just open a connection with the websocket at that URL with the stream name (and maybe the stream name is kind of like the peer ID)

SeeRich, 2024-01-14:
- Yes, I believe that webrtcsrc/sink are the newer rust based plugins.
- I just realized that the gstDecoder has a path for webrtc via the webrtcbin element.
- As for the peer ID, I'm honestly not familiar with WebRTC to answer, but your understanding makes sense to me as well (more related to signalling than the core media transfer).

Unfortunately, I need this to work in a C++/Qt desktop (Windows 11 MSVC) application. Looking at jetson-utils implementation of  the WebRTCServer, this should be do able. I am going to attempt a port this week.

Thanks for the input!