dusty-nv/jetson-utils #203: Saving raw video

topherbuckley, 2024-04-17:
Is there any way to save the raw video rather than a compressed stream using the videoOutput class? I compared the saved video from the `video-viewer csi://0 file://my_video.mp4 --input-codec=raw` against `nvgstcapture-1.0 --mode=2 --automate --capture-auto` and it appears the video-viewer saved video is highly compressed compared to the one from nvgstcapture-1.0. I also tried `video-viewer csi://0 file://my_video.mp4 --input-codec=raw --input-save=input_save.mp4` but get an error from gstreamer "can't use the --input-save option on a raw/uncompressed input stream". I also notice that the stream I see in the X11 window before the output file is saved also looks uncompressed as expected.

dusty-nv, 2024-04-17:
@topherbuckley you can set the output bitrate with `--bitrate` and dump image sequences:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#sequences

topherbuckley, 2024-04-17:
Got it, thank you for the super speedy reply!