dusty-nv/jetson-utils #210: WebRTC getHostByName  not successful

JanGerber, 2024-05-23:
I'am trying to connect to a webrtc stream but just get the initial page:

[image]

And receiving the following error message:

```
(visionary_rtsp:65): GLib-CRITICAL **: 14:44:26.171: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
2024-05-23T14:44:26.193414186Z [webrtc] websocket /color -- recieved message from 192.168.0.3 (peer_id=0) (236 bytes)
2024-05-23T14:44:26.193458891Z [webrtc] received ICE message on /color from 192.168.0.3 (peer_id=0) with mline index 0; candidate: 
2024-05-23T14:44:26.193466027Z candidate:4017222681 1 udp 2113937151 f4cca324-749d-4a5c-854a-6611fa9b2d25.local 55992 typ host generation 0 ufrag QQ7y network-cost 999
2024-05-23T14:44:26.220646474Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 1 of 10)
2024-05-23T14:44:26.225799023Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 2 of 10)
2024-05-23T14:44:26.230952532Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 3 of 10)
2024-05-23T14:44:26.236209596Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 4 of 10)
2024-05-23T14:44:26.240924662Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 5 of 10)
2024-05-23T14:44:26.245598703Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 6 of 10)
2024-05-23T14:44:26.250941369Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 7 of 10)
2024-05-23T14:44:26.255406573Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 8 of 10)
2024-05-23T14:44:26.260139271Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 9 of 10)
2024-05-23T14:44:26.265223754Z [network] getHostByName() trying to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local' (retry 10 of 10)
2024-05-23T14:44:26.265241515Z [network] getHostByName() unable to resolve host 'f4cca324-749d-4a5c-854a-6611fa9b2d25.local'
2024-05-23T14:44:26.265249323Z [webrtc] couldn't resolve f4cca324-749d-4a5c-854a-6611fa9b2d25.local from SDP candidate string
2024-05-23T14:44:26.265467473Z 
2024-05-23T14:44:26.265483345Z (visionary_rtsp:65): GLib-CRITICAL **: 14:44:26.265: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
```
Any idea how to fix this problem

dusty-nv, 2024-05-23:
@JanGerber can you try disabling `chrome://flags/#enable-webrtc-hide-local-ips-with-mdns` ?

JanGerber, 2024-05-23:
@dusty-nv Thank you very much for the very quick response. The disabling was the fix

closed 2024-05-23 by JanGerber