dusty-nv/jetson-utils #211: requesting bindings for `TextExtents`

teresi, 2024-05-29:
Could the bindings for `cudaFont.TextExtents` be added?

I see that `cudaFont` has `GetSize` and `OverlayText` exposed, but not `GetExtents`

https://github.com/dusty-nv/jetson-utils/blob/master/cuda/cudaFont.h

```
class cudaFont(builtins.object)
 |  Bitmap font overlay rendering with CUDA
 |
 |  Methods defined here:
 |
 |  GetSize(...)
 |      Return the size of the font (height in pixels)
 |
 |  OverlayText(...)
 |      Render the font overlay for a given text string
 |
 |  __init__(self, /, *args, **kwargs)
 |      Initialize self.  See help(type(self)) for accurate signature.
 |
```

I'm trying to debug an issue where I get 

```
[cuda]   cudaDeviceSynchronize()
[cuda]      an illegal memory access was encountered (error 700) (hex 0x2BC)
[cuda]      /jetson-inference/utils/codec/gstEncoder.cpp:717
```
```
Exception ignored in: 'cupy.cuda.function.Module.__dealloc__'
```

I think this happens when I use `jetson_utils.cudaFont.OverlayText` and `yolov8`

and I'm looking into creating a minimum working example for that.

```
    return torch.ops.torchvision.nms(boxes, scores, iou_threshold)
  File "/usr/local/lib/python3.10/dist-packages/torch/_ops.py", line 854, in __call__
    return self_._op(*args, **(kwargs or {}))
RuntimeError: CUDA error: an illegal instruction was encountered
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
```

```
  File "cupy_backends/cuda/api/runtime.pyx", line 144, in cupy_backends.cuda.api.runtime.check_status
cupy_backends.cuda.api.runtime.CUDARuntimeError: cudaErrorIllegalInstruction: an illegal instruction was encountered

```

I'm wondering if I can prevent the `cudaDeviceSynchronize` if I don't draw when it get's too close to an edge.

I can rail the top left position, and if I could find the size of the overlay I could skip the overlay if it would go out of bounds on the bottom right position.

I understand that this may not be the fix but I'm looking to experiment.

I'm on jetson_utils 1.0.0

```
>>> import jetson_utils
>>> jetson_utils.VERSION
'1.0.0'
```

dusty-nv, 2024-06-17:
Hi @teresi, sorry for the delay, this was added in https://github.com/dusty-nv/jetson-utils/commit/4152d8acbaa7fc867521d1e7dc4e850f5a28b364

teresi, 2024-06-18:
much appreciated, thank you

closed 2024-06-18 by teresi