dusty-nv/jetson-utils #212: Newly added CUDA stream build errors

champred, 2024-06-19:
https://github.com/dusty-nv/jetson-utils/blob/09060ac8516b117d97c23658bcc6b728ec68904d/python/bindings/PyCUDA.cpp#L1422

This line is missing the third parameter. According to [this documentation](https://docs.nvidia.com/cuda/cuda-runtime-api/group__CUDART__STREAM.html#group__CUDART__STREAM_1g7840e3984799941a61839de40413d1d9) it is optional, but some versions of CUDA may not have that causing compilation to fail. For safety this should be specified as 0.

Additionally the `PYLONG_FROM_PTR` macro is not defined for Python 2 in `PyUtils.h` meaning that the build will fail on that as well.

dusty-nv, 2024-06-20:
Thanks @champred , fixed this in https://github.com/dusty-nv/jetson-utils/commit/903fae30ecafaa5ecdfc8e34f4dec2025ad31778 and confirmed that it is building/working again on JetPack 4 👍

champred, 2024-06-21:
Awesome thank you!

closed 2024-06-21 by champred