dusty-nv/jetson-utils #214: videoOutput Rtsp Memory leak

D3Nd3R, 2024-09-04:
### ENV

jetson agx orin
jetpack 5.1
cuda 11.4.315

### Description

Memory leak when calling video Output::Render. But the memory stops increasing when I connect to the stream via ffplay. After disconnecting, the memory also does not increase

### Step to reproduce

start video-viewer
```sh
./video-viewer rtsp://192.168.1.149:8560/downtown rtsp://192.168.1.202:8554/front --input-codec=h264 --input-decoder=v4l2 --output-codec=h265 --output-encoder=v4l2 --headless
```

run jtop and watch memory usage. It will be increase

run ffplay
```sh
ffplay -fflags nobuffer -flags low_delay -framedrop -avioflags direct rtsp://192.168.1.202:8554/fron
```

memory increase stops.

D3Nd3R, 2024-09-04:
possible work around:

```cpp
   if (_videoOutput->Open())
    {
        cv::VideoCapture capture;
        capture.open(_url);
        if (capture.isOpened())
            capture.release();
    }
    ```

lin-voxvision, 2024-09-10:
same issue, hope can be resolved soon

D3Nd3R, 2024-10-29:
another workaround

_url - is rtsp server url

```cpp
void RtspJetsonUtil::WorkaroundLoop()
{
    const std::string cmd { "ffprobe " + _url + " > /dev/null 2>&1" };
    constexpr std::chrono::seconds sleepTime { 10 };

    while (WorkerBase::IsRunning())
    {
        const auto res { std::system(cmd.c_str()) };
        std::this_thread::sleep_for(sleepTime);

        if (!!res)
            std::cerr << "Command " << cmd << " finished with " << res << std::endl;
    }
}
```