dusty-nv/jetson-utils #236: Latest jetson-utils fails to build

denzilferreira, 2025-04-07:
Had to checkout commit 6d5471c97dad443776d2339569a4d1145f034a9e in order to build a container with jetson-utils.

Several python modules are missing: tabulate, termcolor, docker for some reason :/

D

Birkenpapier, 2025-05-27:
Yes, for me its 'termcolor' which breaks the fun 
Any solution for this except pulling older commits?

dusty-nv, 2025-05-27:
Hi @Birkenpapier @denzilferreira, apologies for the delay - add one of these pip install commands to update jetson-utils:

https://github.com/dusty-nv/jetson-utils?tab=readme-ov-file#install-with-pip-python-only

The dependencies can be found in https://github.com/dusty-nv/jetson-utils/blob/master/pyproject.toml

The prior branch you can roll back to is `v1`:  https://github.com/dusty-nv/jetson-utils/tree/v1

justincdavis, 2025-06-10:
@Birkenpapier Unsure if you are still having issues, but you may need to upgrade your version of pip.

`python3 -m pip install --upgrade pip`

That may also resolve those PEP660 complaints you mention.

closed 2026-03-14 by denzilferreira